ul-gaul/Avionique_Software #22: Paramètres du port série configurable

MaitreMenard, 2017-10-17:
Il serait utile de pouvoir modifier les paramètres du port série (baudrate, start byte, etc.) dans un menu de paramètres du logiciel au lieu de devoir faire ces modifications dans le code source.

konoufo, 2019-03-03:
Il faut considérer que certains paramètres importants doivent être validés avant la sauvegarde (du menu des paramètres) et avant l'initialisation des contrôleurs. Vu que ces paramètres sont connus on peut coder la validation dans le contrôleur du menu des paramètres et, en cas d'erreur, rétroagir dans l'UI du menu des paramètres.

konoufo, 2019-03-03:
Si on veut que la sauvegarde ait un effet dynamique sur l'application en cours (affecter le dessin de la courbe altitude quand on change `target_altitude`), on devrait permettre à l'application (contrôleurs de l'application) de pouvoir s'abonner (register) aux changements dans la configuration. En gros: `Controller` **registers**  to changes in `Config` (datastore object) which **registers** to changes in `ConfigController` (contrôleur du menu des paramètres).